NeuronRobotics/nrjavaserial #191: dlopen failed: cannot find verneed/verdef

doit-ceo, 2020-06-13:
I'm trying to build apk for Android x86_64 

After I faced problem here #190  I managed to copy the libNRJavaSerial.so in to /main/java/jiniLibs/ and managed to load the lib by System.setProperty("libNRJavaSerial.userlib", "sys"); 

First there is bug in NativeResource, line 76, this should remove the "lib" (as SerialManager have it as "libNRJavaSerial" and System.loadLibrary will always adds "lib" to the name). So I had to rename my file to "liblibNRJavaSerial" to make it work with System.setProperty("libNRJavaSerial.userlib", "sys"); 

Furthermore, I had to load (after copying them to "/main/java/jiniLibs/" from my Linux ) System.loadLibrary("ld-linux-x86-64") and System.loadLibrary("c") as it kept requesting it, but then I got the final error :

```
java.lang.UnsatisfiedLinkError: dlopen failed: cannot find verneed/verdef for version index=32770 referenced by symbol "_res" at "/data/app/~~V7806S1Z6o573KkQbjE8XQ==/sa.com.doit.appx.pt-s5427NUiFM0BAaVlWkBI9A==/lib/x86_64/libc.so"

```

What am I doing wrong ?!

madhephaestus, 2020-06-14:
The standard linux x86 binary should work for this, and Android is just linux.

closed 2020-06-14 by madhephaestus

doit-ceo, 2020-06-15:
Actually this never worked until I compiled it myself with Android NDK with this for future reference, yet this was valid lib but not linked correctly with the java jar (then I revert to using the original rxtx jar)

```
rm -R /home/xxx/work/yyy/dev/rxtx/nrjavaserial/src/main/c/build
mkdir /home/xxx/work/yyy/dev/rxtx/nrjavaserial/src/main/c/build/
mkdir /home/xxx/work/yyy/dev/rxtx/nrjavaserial/src/main/c/build/bin

CLANG=/home/xxx/Android/Sdk/ndk/21.3.6528147/toolchains/llvm/prebuilt/linux-x86_64/bin/x86_64-linux-android30-clang
INCLUDE='-I /usr/lib/jvm/java-1.8.0-openjdk-amd64/include/linux/ -I /usr/lib/jvm/java-1.8.0-openjdk-amd64/include/ -I include -I include/target'
ARGS='-O3 -Wall -fmessage-length=0 -fPIC -m64 -MMD'
ARGSX='-m64 -shared'

$CLANG -target x86_64-linux-android $INCLUDE $ARGS -c src/fixup.c -o build/fixup.o 
$CLANG -target x86_64-linux-android $INCLUDE $ARGS -c src/fuserImp.c -o build/fuserImp.o
$CLANG -target x86_64-linux-android $INCLUDE $ARGS -c src/SerialImp.c -o build/SerialImp.o
echo "********LINKING*******"

$CLANG -nostartfiles build/fixup.o build/fuserImp.o build/SerialImp.o -lm -pthread $ARGSX -o "/home/xxx/work/yyy/dev/rxtx/nrjavaserial/src/main/c/build/bin/libNRJavaSerial.so" 
```