NeuronRobotics/nrjavaserial #192: Android Support

doit-ceo, 2020-06-14:
OK , after 3 days, I finally managed to get this compiled with NDK, the jar didn't work (native biding never worked in Android with Jar) , then I used the legacy RXTX jar, this seems to get at least the native version printed, but nothing else work !

Does this lib really support Android, if so, can someone help in this ?

madhephaestus, 2020-06-14:
It only works on *rooted* Android builds since Android 3.x

The linux kernel under android mounts serial ports with permission

```
root:root
```

There are 2 ways to allow the Dalvik system to access that, either run the application as root, or modify the sub system with a udev rule telling the kernel to mount the devices as owned by some some group the user is already a member of. 

Basically, the easy use case was removed in Android 4.x

Moding forward the USB api is recommended for accessing usb connected serial devices. You can ask for permission to access usb devices from Android.

closed 2020-06-14 by madhephaestus

doit-ceo, 2020-06-15:
@madhephaestus thanks for your replies, I do confirm that my device is rooted, yet this doesn't work ! and we have a call for root just before rxtx calls

```
        try {
            Process px = Runtime.getRuntime().exec("su");
        } catch (Exception e) {
            e.printStackTrace();
        }
        //rxtx here 
```

madhephaestus, 2020-06-15:
rooting is step one.

Your application needs to be run as root, or you need to mount the serial port with a udev rule mounting serial ports as 777

Open a root terminal and do ls -al on the port you want to access. 

Make a udev rule to mount the serial port with permissions 777

```
SUBSYSTEM=="tty", MODE="0777"
```