NeuronRobotics/nrjavaserial #193: Replace System.out/System.err printing with SLF4J

MrDOS, 2020-06-16:
There are a number of places where the library blasts some output directly to stdout or stderr, which is quite impolite for a library. And there are other places where it might be nice to have more debugging statements, but it is impractical to add them given the current inability of end users to suppress their output. @madhephaestus, any objection to me adding a dependency on the SLF4J API and replacing all the current direct output calls with proper logging?

madhephaestus, 2020-06-16:
great idea. Something to concider, is there a way to allow the C code to print to Java through the logger? the C level prints were never a good idea

MrDOS, 2020-06-16:
I think that's possible. It's certainly possible to call back into Java from JNI code, so as long as the logger can be accessed consistently (e.g., a static `log` field on the class, as is idiomatic anyway), I think I can rework the existing `report_verbose`/`report_error`/`report_warning`/`report` functions to call SLF4J.

As I mentioned in #189, I think it's probably more important/I'm more interested in tackling the problem of port locking and its portability. But this will be rattling around in the back of my mind.