NeuronRobotics/nrjavaserial #207: Bump apache.commons.net to 3.7.2 to fix NPEs

t2000, 2021-01-22:
See https://github.com/openhab/openhab-core/issues/2140

```
java.lang.NullPointerException: null
        at org.apache.commons.net.telnet.TelnetClient._closeOutputStream(TelnetClient.java:84) ~[?:?]
        at org.apache.commons.net.telnet.TelnetOutputStream.close(TelnetOutputStream.java:163) ~[?:?]
        at org.apache.commons.net.telnet.TelnetClient.disconnect(TelnetClient.java:129) ~[?:?]
        at gnu.io.rfc2217.TelnetSerialPort.close(TelnetSerialPort.java:321) ~[?:?]
        at org.openhab.core.io.transport.serial.rxtx.RxTxSerialPort.close(RxTxSerialPort.java:51) ~[?:?]
```

This NPE has been fixed in 3.7.2, so could you please bump up the version, here? 

https://github.com/NeuronRobotics/nrjavaserial/blob/6a97d715333cf73ffbf97a218438b916fb9dcc7c/build.gradle#L45

MrDOS, 2021-01-22:
Sure thing. It may be a little while before I can cut a new release, however. If you're in a hurry, you use the JAR [from this build action](https://github.com/NeuronRobotics/nrjavaserial/actions/runs/503595296) ([nrjavaserial-4583df9](https://github.com/NeuronRobotics/nrjavaserial/suites/1879653918/artifacts/36682314) under the “Artifacts” header).

closed 2021-01-22 by MrDOS

t2000, 2021-01-22:
@MrDOS Awesome, thank you very much for the quick fix.

Looking forward to get a new release from nrjavaserial soon!

fwolter, 2021-01-24:
Maybe it's a good idea to release a new version before #194 is merged? I think the slf4j logging could introduce some instability and needs thoroughly testing, which could take some time.

MrDOS, 2021-01-24:
I spoke with @madhephaestus earlier this week about his vision for the project. Due to the fact none of his other projects depend on NRJavaSerial any more, he offered to let me take over ongoing maintenance – that's been the basis on which I've started to merge PRs. However, we still haven't completed the transfer of Maven publishing permissions yet, so I can't cut a release (beyond tagging the version in Git). For now, I've spun off a `dev` branch, and will merge #194 and any subsequent major changes into that. As soon as I can, I'll make a new point release (v5.2.2) from `master`; and once we're ready to go with more major changes, we'll release v5.3.0 with the changes from `dev`.

fwolter, 2021-01-24:
Sounds good! I love to see this project making progress!

I would file a PR applying the formatter against the dev branch and then continue to improve the logging further.

PR #208 (2021-01-22): Bump Apache Commons Net to v3.7.2
Closes #207.