NeuronRobotics/nrjavaserial #212: After unplug USB cable, java program crashed.

fy-create, 2021-03-01:
Using latest 5.2.1 on MacOS, connected with USB serial port.
after unplug USB cable, java program crashed. 

`
 A fatal error has been detected by the Java Runtime Environment:

  SIGSEGV (0xb) at pc=0x0000000135680acd, pid=28481, tid=0x000000000000c103

 JRE version: OpenJDK Runtime Environment (8.0) (build 1.8.0-adoptopenjdk-jenkins_2018_05_19_02_01-b00)
 Java VM: OpenJDK 64-Bit Server VM (25.71-b00 mixed mode bsd-amd64 compressed oops)
 Problematic frame:
 C  [libNRJavaSerial.jnilib+0x1acd]  drain_loop+0x4d

 Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again

 If you would like to submit a bug report, please visit:
   http://bugreport.java.com/bugreport/crash.jsp

---------------  T H R E A D  ---------------

Current thread is native thread

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x000070000e0f2838

Registers:
RAX=0x0000000000000000, RBX=0x000070000e0f27a0, RCX=0x000070000e1760ac, RDX=0xffffffffffffffff
RSP=0x000070000e175fa0, RBP=0x000070000e175fb0, RSI=0x0000000000000000, RDI=0x000000000000003c
R8 =0x0000000000000001, R9 =0x0000000000000000, R10=0x0000000000000001, R11=0x0000000000000247
R12=0x0000000000000000, R13=0x0000000000000000, R14=0x0000000100000000, R15=0x0000000000000000
RIP=0x0000000135680acd, EFLAGS=0x0000000000010206, ERR=0x0000000000000004
  TRAPNO=0x000000000000000e

Top of Stack: (sp=0x000070000e175fa0)
0x000070000e175fa0:   000070000e176000 0000000000000000
0x000070000e175fb0:   000070000e175fd0 00007fff20594954
0x000070000e175fc0:   0000000000000000 0000000000000000
0x000070000e175fd0:   000070000e175ff0 00007fff205904a7
0x000070000e175fe0:   0000000000000000 0000000000000000
0x000070000e175ff0:   0000000000000000 0000000000000000
0x000070000e176000:   a5ecaf4c39eb0cf3 0000000000000000
0x000070000e176010:   000070000e279000 000070000deed010
0x000070000e176020:   0000000000000000 0000000000000000
0x000070000e176030:   0000000000000001 0000000a0000001f
0x000070000e176040:   0000000000000000 0001000000000000
0x000070000e176050:   0000000000000000 0000000000000000
0x000070000e176060:   0000000000000000 0000000000000000
0x000070000e176070:   0000000000000000 0000000000000000
0x000070000e176080:   0000000000000000 0000000000000000
0x000070000e176090:   0000000135680a80 000070000e0f27a0
0x000070000e1760a0:   0003000000000000 0000003c00000000
0x000070000e1760b0:   000070000e176000 000070000e0f6000
0x000070000e1760c0:   000070000e0f5000 0000000000083000
0x000070000e1760d0:   0000000000001000 00000000000796a3
0x000070000e1760e0:   000070000e176000 000070000e1760ac
0x000070000e1760f0:   0000000000000000 000000000000c103
0x000070000e176100:   00000000000008ff 0000000000000000
0x000070000e176110:   0000000000000000 a5ecdf4c37fc6cf3
0x000070000e176120:   0000000000000000 0000000000000000
0x000070000e176130:   0000000000000000 0000000000000000
0x000070000e176140:   0000000000000000 0000000000000000
0x000070000e176150:   0000000000000000 0000000000000000
0x000070000e176160:   0000000000000000 0000000000000000
0x000070000e176170:   0000000000000000 0000000000000000
0x000070000e176180:   0000000000000000 0000000000000000
0x000070000e176190:   0000000000000000 0000000000000000 

Instructions: (pc=0x0000000135680acd)
0x0000000135680aad:   1f 40 00 e8 35 62 00 00 83 38 04 75 40 83 bb 98
0x0000000135680abd:   00 00 00 00 75 37 bf 40 42 0f 00 e8 5b 63 00 00
0x0000000135680acd:   83 bb 98 00 00 00 00 75 24 8b 3b e8 27 63 00 00
0x0000000135680add:   85 c0 75 cf 83 bb 6c 01 00 00 00 74 d0 4c 89 b3 

Register to memory mapping:

RAX=0x0000000000000000 is an unknown value
RBX=0x000070000e0f27a0 is an unknown value
RCX=0x000070000e1760ac is an unknown value
RDX=0xffffffffffffffff is an unknown value
RSP=0x000070000e175fa0 is an unknown value
RBP=0x000070000e175fb0 is an unknown value
RSI=0x0000000000000000 is an unknown value
RDI=0x000000000000003c is an unknown value
R8 =0x0000000000000001 is an unknown value
R9 =0x0000000000000000 is an unknown value
R10=0x0000000000000001 is an unknown value
R11=0x0000000000000247 is an unknown value
R12=0x0000000000000000 is an unknown value
R13=0x0000000000000000 is an unknown value
R14=0x0000000100000000 is an unknown value
R15=0x0000000000000000 is an unknown value

Stack: [0x000070000e0f6000,0x000070000e176000],  sp=0x000070000e175fa0,  free space=511k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libNRJavaSerial.jnilib+0x1acd]  drain_loop+0x4d
C  [libsystem_pthread.dylib+0x6954]  _pthread_start+0xe0
C  [libsystem_pthread.dylib+0x24a7]  thread_start+0xf`

MrDOS, 2021-03-01:
Thanks for the bug report. This has previously been discussed in #197, and is partially fixed in #211; however, that fix causes a different crash on Windows that I haven't tracked down yet.

If you can do without the `HARDWARE_ERROR` serial port event feature added in v5.1.0, then you can try using [v5.0.2](https://mvnrepository.com/artifact/com.neuronrobotics/nrjavaserial/5.0.2), the newest version which does not have this bug.

fy-create, 2021-03-02:
Thanks, v5.0.2 is OK.
I hope next version can fix this issue, both mac and windows. :)

closed 2021-03-02 by fy-create