NeuronRobotics/nrjavaserial #232: Compiling Problem

celioregoli, 2023-04-05:
celio@Lenovo-G505:~/nrjavaserial$  sudo make -C src/main/c crosstools && make arm
make: Entering directory '/home/celio/nrjavaserial/src/main/c'
JAVA_HOME.mk:112: The JAVA_HOME environment variable has not been defined.
Based on the location of the `java` binary, I've guessed it's:

    "/usr/lib/jvm/java-20-openjdk-amd64"

If that's not correct, or to suppress this message, explicitly set the
JAVA_HOME environment variable to the location of your preferred Java
installation.
apt install --install-recommends \
            gcc-i686-linux-gnu \
            gcc-arm-linux-gnueabi \
            gcc-arm-linux-gnueabihf \
            gcc-aarch64-linux-gnu \
            gcc-mingw-w64 \
            gcc-powerpc-linux-gnu
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gcc-i686-linux-gnu is already the newest version (4:12.2.0-1ubuntu1).
gcc-arm-linux-gnueabi is already the newest version (4:12.2.0-1ubuntu1).
gcc-arm-linux-gnueabihf is already the newest version (4:12.2.0-1ubuntu1).
gcc-aarch64-linux-gnu is already the newest version (4:12.2.0-1ubuntu1).
gcc-mingw-w64 is already the newest version (10.3.0-15ubuntu1+24.4ubuntu1).
gcc-powerpc-linux-gnu is already the newest version (4:12.2.0-1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
make: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -C src/main/c arm
make[1]: Entering directory '/home/celio/nrjavaserial/src/main/c'
JAVA_HOME.mk:112: The JAVA_HOME environment variable has not been defined.
Based on the location of the `java` binary, I've guessed it's:

    "/usr/lib/jvm/java-20-openjdk-amd64"

If that's not correct, or to suppress this message, explicitly set the
JAVA_HOME environment variable to the location of your preferred Java
installation.
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv5.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv6.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv6_HF.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv7.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
arm-linux-gnueabihf-gcc -I""/usr/lib/jvm/java-20-openjdk-amd64"/include" -I./include -I./include/target -O3 -fPIC -c -Wall -I"/usr/lib/jvm/java-20-openjdk-amd64"/include/linux -U_FORTIFY_SOURCE -march=armv7-a src/fixup.c -o build/linux/ARM_32/fixupv7_HF.o
cc1: error: '-mfloat-abi=hard': selected architecture lacks an FPU
make[2]: *** [natives.mk:15: build/linux/ARM_32/fixupv7_HF.o] Error 1
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make[1]: *** [Makefile:172: arm32v7HF] Error 2
make[1]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make: *** [Makefile:15: arm] Error 2
celio@Lenovo-G505:~/nrjavaserial$ export LC_ALL=C  sudo make -C src/main/c crosstools && make arm
bash: export: `-C': not a valid identifier
bash: export: `src/main/c': not a valid identifier
celio@Lenovo-G505:~/nrjavaserial$ export LC_ALL=C make -C src/main/c crosstools && make arm
bash: export: `-C': not a valid identifier
bash: export: `src/main/c': not a valid identifier

MrDOS, 2023-04-05:
This may be related to [Ubuntu bug #1939379](https://bugs.launchpad.net/ubuntu/+source/gcc-defaults/+bug/1939379) (I assume you're using Ubuntu here). Can you try modifying the makefile for the native libraries to use `-march=armv7-a+fp` instead of `-march=armv7-a`?

```diff
diff --git a/src/main/c/Makefile b/src/main/c/Makefile
index d7c5a39..56b2b82 100644
--- a/src/main/c/Makefile
+++ b/src/main/c/Makefile
@@ -164,8 +164,8 @@ arm32v7:
 
 # Requires gcc-arm-linux-gnueabihf.
 arm32v7HF: export CC := arm-linux-gnueabihf-gcc
-arm32v7HF: export CFLAGS += $(LINUX_CFLAGS) -march=armv7-a
-arm32v7HF: export LDFLAGS += -march=armv7-a
+arm32v7HF: export CFLAGS += $(LINUX_CFLAGS) -march=armv7-a+fp
+arm32v7HF: export LDFLAGS += -march=armv7-a+fp
 arm32v7HF: export platform := linux/ARM_32
 arm32v7HF: export variant := v7_HF
 arm32v7HF:
```

celioregoli, 2023-04-06:
thank you very much for the advice, yes , exactly , I have kubuntu 22.10, I did as you said and now it continues without that error, but it gives another one.

sudo make -C src/main/c crosstools && make arm
make: Entering directory '/home/celio/nrjavaserial/src/main/c'
JAVA_HOME.mk:112: The JAVA_HOME environment variable has not been defined.
Based on the location of the `java` binary, I've guessed it's:

    "/usr/lib/jvm/java-20-openjdk-amd64"

If that's not correct, or to suppress this message, explicitly set the
JAVA_HOME environment variable to the location of your preferred Java
installation.
apt install --install-recommends \
            gcc-i686-linux-gnu \
            gcc-arm-linux-gnueabi \
            gcc-arm-linux-gnueabihf \
            gcc-aarch64-linux-gnu \
            gcc-mingw-w64 \
            gcc-powerpc-linux-gnu
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gcc-i686-linux-gnu is already the newest version (4:12.2.0-1ubuntu1).
gcc-arm-linux-gnueabi is already the newest version (4:12.2.0-1ubuntu1).
gcc-arm-linux-gnueabihf is already the newest version (4:12.2.0-1ubuntu1).
gcc-aarch64-linux-gnu is already the newest version (4:12.2.0-1ubuntu1).
gcc-mingw-w64 is already the newest version (10.3.0-15ubuntu1+24.4ubuntu1).
gcc-powerpc-linux-gnu is already the newest version (4:12.2.0-1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
make: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -C src/main/c arm
make[1]: Entering directory '/home/celio/nrjavaserial/src/main/c'
JAVA_HOME.mk:112: The JAVA_HOME environment variable has not been defined.
Based on the location of the `java` binary, I've guessed it's:

    "/usr/lib/jvm/java-20-openjdk-amd64"

If that's not correct, or to suppress this message, explicitly set the
JAVA_HOME environment variable to the location of your preferred Java
installation.
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv5.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv6.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv6_HF.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv7.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv7_HF.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
make[2]: 'resources/native/linux/ARM_32/libNRJavaSerialv8.so' is up to date.
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make -f natives.mk
make[2]: Entering directory '/home/celio/nrjavaserial/src/main/c'
arm-linux-gnueabihf-gcc -I""/usr/lib/jvm/java-20-openjdk-amd64"/include" -I./include -I./include/target -O3 -fPIC -c -Wall -I"/usr/lib/jvm/java-20-openjdk-amd64"/include/linux -U_FORTIFY_SOURCE -march=armv8-a src/fixup.c -o build/linux/ARM_32/fixupv8_HF.o
cc1: error: '-mfloat-abi=hard': selected architecture lacks an FPU
make[2]: *** [natives.mk:15: build/linux/ARM_32/fixupv8_HF.o] Error 1
make[2]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make[1]: *** [Makefile:190: arm32v8HF] Error 2
make[1]: Leaving directory '/home/celio/nrjavaserial/src/main/c'
make: *** [Makefile:15: arm] Error 2

celioregoli, 2023-04-06:
I tried, searching on the internet, to modify
# Requires gcc-arm-linux-gnueabihf.
arm32v8HF: export CC := arm-linux-gnueabihf-gcc
arm32v8HF: export CFLAGS += $(LINUX_CFLAGS) -m=armv8-a
arm32v8HF: export LDFLAGS += -with-arch=armv8-a
arm32v8HF: export platform := linux/ARM_32
arm32v8HF: export variant := v8_HF
arm32v8HF:

with

# Requires gcc-arm-linux-gnueabihf.
arm32v8HF: export CC := arm-linux-gnueabihf-gcc
arm32v8HF: export CFLAGS += $(LINUX_CFLAGS) -m=armv8-a+fp
arm32v8HF: export LDFLAGS += -with-arch=armv8-a+fp
arm32v8HF: export platform := linux/ARM_32
arm32v8HF: export variant := v8_HF
arm32v8HF:

now continues but
give another error

MrDOS, 2023-04-06:
You probably need to add `+fp` to the `-march` flag for the `arm64v8` target on lines 194 and 195, too (i.e., `-march=armv8-a+fp`):

https://github.com/NeuronRobotics/nrjavaserial/blob/e897e3978aba971c43c77ef72cf1e8593376e3b8/src/main/c/Makefile#L194-L195

celioregoli, 2023-04-06:
I had already tried -march=armv8-a+fp and it gave me error, so after searching on the web I put arm32v8HF: export LDFLAGS += -with-arch=armv8-a+fp and it worked.

celioregoli, 2023-04-06:
celio@Lenovo-G505:~/nrjavaserial$ sudo make -C src/main/c crosstools && make arm
make: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
JAVA_HOME.mk:112: The JAVA_HOME environment variable has not been defined.
Based on the location of the `java` binary, I've guessed it's:

    "/usr/lib/jvm/java-20-openjdk-amd64"

If that's not correct, or to suppress this message, explicitly set the
JAVA_HOME environment variable to the location of your preferred Java
installation.
apt install --install-recommends \
            gcc-i686-linux-gnu \
            gcc-arm-linux-gnueabi \
            gcc-arm-linux-gnueabihf \
            gcc-aarch64-linux-gnu \
            gcc-mingw-w64 \
            gcc-powerpc-linux-gnu
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze... Fatto
Lettura informazioni sullo stato... Fatto   
gcc-i686-linux-gnu è già alla versione più recente (4:12.2.0-1ubuntu1).
gcc-arm-linux-gnueabi è già alla versione più recente (4:12.2.0-1ubuntu1).
gcc-arm-linux-gnueabihf è già alla versione più recente (4:12.2.0-1ubuntu1).
gcc-aarch64-linux-gnu è già alla versione più recente (4:12.2.0-1ubuntu1).
gcc-mingw-w64 è già alla versione più recente (10.3.0-15ubuntu1+24.4ubuntu1).
gcc-powerpc-linux-gnu è già alla versione più recente (4:12.2.0-1ubuntu1).
0 aggiornati, 0 installati, 0 da rimuovere e 29 non aggiornati.
make: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -C src/main/c arm
make[1]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
JAVA_HOME.mk:112: The JAVA_HOME environment variable has not been defined.
Based on the location of the `java` binary, I've guessed it's:

    "/usr/lib/jvm/java-20-openjdk-amd64"

If that's not correct, or to suppress this message, explicitly set the
JAVA_HOME environment variable to the location of your preferred Java
installation.
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_32/libNRJavaSerialv5.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_32/libNRJavaSerialv6.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_32/libNRJavaSerialv6_HF.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_32/libNRJavaSerialv7.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_32/libNRJavaSerialv7_HF.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_32/libNRJavaSerialv8.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_32/libNRJavaSerialv8_HF.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make -f natives.mk
make[2]: ingresso nella directory «/home/celio/nrjavaserial/src/main/c»
make[2]: «resources/native/linux/ARM_64/libNRJavaSerialv8.so» è aggiornato.
make[2]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
make[1]: uscita dalla directory «/home/celio/nrjavaserial/src/main/c»
./gradlew build

FAILURE: Build failed with an exception.

* What went wrong:
Could not initialize class org.codehaus.groovy.runtime.InvokerHelper
> Exception java.lang.NoClassDefFoundError: Could not initialize class org.codehaus.groovy.reflection.ReflectionCache [in thread "Daemon worker"]

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s
make: *** [Makefile:16: arm] Errore 1
celio@Lenovo-G505:~/nrjavaserial$

celioregoli, 2023-04-07:
i solved, i had java 17 , i put java 11 and all went well.
Thanks for your suggestions.
Problem Solved

closed 2026-09-08 by madhephaestus