NeuronRobotics/nrjavaserial #240: Apple M1

mothattack, 2023-08-29:
I have not been able to build the rxtx files for the Apple M1, only the old intel macs. Has anyone had any luck?

MrDOS, 2023-08-29:
We've discussed it a few times: https://github.com/NeuronRobotics/nrjavaserial/issues/195, https://github.com/NeuronRobotics/nrjavaserial/issues/219. And I revisited and revised [my branch](https://github.com/MrDOS/nrjavaserial/tree/build-universal-macos-binary) just last week. If you're looking for a working binary, you can grab the artifact from [the latest build from that branch](https://github.com/MrDOS/nrjavaserial/actions/runs/5978999990).

My branch is currently based on https://github.com/NeuronRobotics/nrjavaserial/pull/238. I plan to PR'ify it after that other PR has merged.

mothattack, 2023-08-30:
excellent - thank you!

closed 2026-09-08 by madhephaestus