NeuronRobotics/nrjavaserial #243: Two years without a release, any plan for a new release?

sblantipodi, 2023-09-05:
As title :)

MrDOS, 2023-09-05:
Sorry, saw [your comment](https://github.com/NeuronRobotics/nrjavaserial/pull/241#issuecomment-1704288492) and hadn't gotten around to replying yet.

Per [the v5.3.0 milestone](https://github.com/NeuronRobotics/nrjavaserial/milestone/2), there are a few showstopping stability/crash issues that I want to resolve before we push out another release. In particular, the issue where a hardware error on macOS causes a JVM crash (https://github.com/NeuronRobotics/nrjavaserial/pull/211). I'm revisiting that fix and hope to have a solution soon.

sblantipodi, 2023-09-05:
thank you very much for the answer :) I appreciate it...
keep up the good work 👍

closed 2023-09-05 by sblantipodi