NeuronRobotics/nrjavaserial #252: Is the project abandoned?

sblantipodi, 2024-07-05:
as title.
Is the project abandoned?

andyrozman, 2024-07-05:
Good question... There hasn't been much activity here... but I have been using library for a quite long time now and it is working pretty stable (using on linux mostly for older diabetic devices), but number of serial devices seems to be dwindiling... 
There are some forks it seems, but they don't seem to have much activity either.

MrDOS, 2024-07-05:
I'd say “starved for attention” before I'd say “abandoned”. While the library is feature-sufficient for many users, there are several improvements I'd like to see (e.g., better port detection on Windows without JNA; moving more functionality to native Java to reduce the size of native libraries) in addition to support for additional platforms (e.g., RISC-V Linux). However, at present, the project has fairly different code paths across Windows, macOS, and *nix, particularly around opening/closing ports; this means that even small changes come with a high risk of introducing subtle bugs.

Manually building for and testing across the wide array of supported platforms is a tedious, fraught process. In my mind, automated building (https://github.com/NeuronRobotics/nrjavaserial/pull/238) and testing (https://github.com/NeuronRobotics/nrjavaserial/pull/182) need to happen before I can think about anything else. That CI issue is all tee'd up for me to put a bit more time into it, and I think about it every time a long weekend rolls around. Unfortunately, I have some fairly major Real Life commitments coming up over the next few months, and I doubt I'll personally have time to look at it before early winter.